acuparse/acuparse #15: PWSweather Updates not working

mfrazzz, 2018-01-13:
Updates going to pwsweather do not seem to be working.  In syslog I can see the update trying but there is never a response back from pwsweather:

`Jan 13 08:40:01 cygnus acuparse(2.1.8): PWS Query: &dateutc=2018-01-13+15:40:01&tempf=32.4&winddir=270&windspeedmph=0&baromin=30.58&humidity=63&dewptf=21&rainin=0&dailyrainin=0 | Result:
Jan 13 08:41:02 cygnus acuparse(2.1.8): PWS Query: &dateutc=2018-01-13+15:41:02&tempf=32.6&winddir=270&windspeedmph=0&baromin=30.58&humidity=61&dewptf=20.4&rainin=0&dailyrainin=0 | Result:
Jan 13 08:42:01 cygnus acuparse(2.1.8): PWS Query: &dateutc=2018-01-13+15:42:01&tempf=32.9&winddir=248&windspeedmph=0&baromin=30.58&humidity=60&dewptf=20.4&rainin=0&dailyrainin=0 | Result:`

I also tried to do this interactively in the browser using the format that is constructed:

`pwsweather.com/pwsupdate/pwsupdate.php?ID=KCOFALCO8&password=xxxxxxx&dateutc=2018-01-13+15:40:01&tempf=32.4&winddir=270&windspeedmph=0&baromin=30.58&humidity=63&dewptf=21&rainin=0&dailyrainin=0`

Which gives back a page with the following text:

Data posted for Station KCOFALCO8 on Jan 13, 2018

You can always find the most recent posting of your data at http://www.hamweather.net/weather/current/KCOFALCO8.html

No Data posted as of yet on Jan 13, 2018

Let me know if there is anything else I can provide.

maxwellpower, 2018-01-15:
Initial investigation leads me to believe something might be blocked on your side, preventing the data from reaching PWS from your system. The code for PWS and for WU are the same, aside from the URL and some minor differences in formatting. If WU works, then, in theory, PWS should also work.

Can you try to wget the update URL from the command line and see what it returns to you? A totally empty response definitely indicates some form of communication issue.

mfrazzz, 2018-01-15:
So there is an issue.  Getting Connection refused via wget:

mfrazier@cygnus ~ $ wget "http://pwsweather.com/pwsupdate/pwsupdate.php?ID=KCOFALCO8&PASSWORD=xxxxxxxx&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0"
--2018-01-15 12:12:41--  http://pwsweather.com/pwsupdate/pwsupdate.php?ID=KCOFALCO8&PASSWORD=juxtapose&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0
Resolving pwsweather.com (pwsweather.com)... 52.73.22.124, 34.199.250.29
Connecting to pwsweather.com (pwsweather.com)|52.73.22.124|:80... failed: Connection refused.
Connecting to pwsweather.com (pwsweather.com)|34.199.250.29|:80... failed: Connection refused.
mfrazier@cygnus ~ $

I've confirmed that the station id and the password are correct.  I've just sent a support request to pwsweather hoping they can find something with my account / station.

maxwellpower, 2018-01-15:
Using your query from my production box, I get the following response:
```
wget "http://pwsweather.com/pwsupdate/pwsupdate.php?ID=KCOFALCO8&PASSWORD=xxxxxxxx&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0"
--2018-01-15 12:27:16--  http://pwsweather.com/pwsupdate/pwsupdate.php?ID=KCOFALCO8&PASSWORD=xxxxxxxx&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0
Resolving pwsweather.com (pwsweather.com)... 34.199.250.29
Connecting to pwsweather.com (pwsweather.com)|34.199.250.29|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 114 [text/html]
Saving to: ‘pwsupdate.php?ID=KCOFALCO8&PASSWORD=xxxxxxxx&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0’

pwsupdate.php?ID=KC 100%[=====================>]     114  --.-KB/s   in 0s

2018-01-15 12:27:16 (31.5 MB/s) - ‘pwsupdate.php?ID=KCOFALCO8&PASSWORD=xxxxxxxx&dateutc=2018-01-15+18:47:01&tempf=11.4&winddir=338&windspeedmph=13&baromin=30.16&humidity=86&dewptf=8.1&rainin=0&dailyrainin=0’ saved [114/114]
```
```
cat pwsupdate.php\?ID\=KCOFALCO8\&PASSWORD\=xxxxxxxx\&dateutc\=2018-01-15+18\:47\:01\&tempf\=11.4\&winddir\=338\&windspeedmph\=13\&baromin\=30.16\&humidity\=86\&dewptf\=8.1\&rainin\=0\&dailyrainin\=0

<html>
<head>
   <title>PWS Weather Station Update</title>
</head>
<body>
ERROR: Not a vailid Station ID/Password
```
Since your password is all X's, obviously the actual update would fail. Something on your network or their side is blocking your connection to them.

mfrazzz, 2018-01-15:
This has been resolved.  I was able to identify that it was only this system on my network having an issue, and then I remembered this system is running PeerGuardian, so it was identifying pwsweather.com's IPs and was blocking connections to them (no idea why those are identified).  I put a whitelist in for the IPs and now my data is working fine (just like updates for WU and CWOP have been working the whole time).  Thanks for the support.  I'll close this issue now.

closed 2018-01-15 by mfrazzz