acuparse/acuparse #19: Block access from Acurite

jemiller0, 2018-02-22:
I haven't yet installed Acuparse, but, hope to probably this weekend. So, this is more of a question that an issue. The question is, does Acuparse have a way to block access to/from Acurite's servers? As you are probably aware, Acurite is stopping their service for SmartHubs in August. My fear is that they could install a firmware update to disable the SmartHub itself (of course it just occurred to me that they could have already put a time bomb in it). Personally, I would like to switch to using something like Acuparse where I have more control. I don't want to be forced into upgrading as I have 20 sensors and would need to buy extra hubs now that Acurite lowered the number of supported sensors from 10 to 7.

maxwellpower, 2018-02-22:
Very shortly I will update the hub portion to stop sending updates to MyAcurite. You can already do this with a select option in the config. Selecting that will not send any updates to MyAcurite and will block firmware updates. The problem is with the firmware for sure. So I will make sure Acuparse locks in the 224 version number we are currently on, even if you are sending updates. This should hopefully prevent them from bricking our hubs.

Orion231, 2018-02-23:
Is the code ready to be pulled for update?

Thanks...

maxwellpower, 2018-02-23:
Not just yet. Will be included in version 2.2.1. Most likely later on this evening. There were some other issues that popped up.

closed 2018-02-24 by maxwellpower

jemiller0, 2018-02-24:
Thanks. And thanks for creating this project to begin with.

commit 6f13d62b (2018-02-23): Prevent SmartHUB firmware updates

Acuparse now sends back it's own response to the HUB.

Resolves #19
commit 3c82cf4a (2018-02-24): Bug Fixes

Fixes issues with MyAcuRite certificates - Resolves #22
Don't send updates after HUB EoL
Use the right update URL when saving settings - Resolves #19