acuparse/acuparse #24: Acuparse Stand Alone Rain Gauge.

21tuxedo, 2018-02-27:
There is presently no support for a Stand Alone AcuRite Rain Gauge. 
I have one presently working with my smartHUB and could provide any data you need.

maxwellpower, 2018-02-27:
This will most likely find it's way into Version 3 with Atlas support. Maybe sooner depending on how long they take with the Atlas. I'm currently playing with new database designs and with how multiple sensors can be better supported.

To support that sensor, all I would really need is the data string that it produces.

21tuxedo, 2018-02-27:
Data from Rain Gauge sent to you. Thanks

maxwellpower, 2018-02-28:
So the main stuff we need is:
- `mt=rain899&sensor=xxxxxxx`
- `rainin=0.00&dailyrainin=0.00&baromin=30.25`

Can you tell me more about why you would need this sensor? Is it used to build the main dashboard or show similar to a tower?

Just trying to think about how it will all pull together for the next version and what the need for this really is. Considering the Access/Atlas roadmap.