acuparse/acuparse #26: Does Access integration mean data doesn’t get uploaded to Acurite?

bdwilson, 2018-03-01:
Curious if your dns interception of the Access API host means that data isn’t delivered to their servers. I’d like to still use their app, but also keep data local. Is this possible? 

Thanks for your work on this!

maxwellpower, 2018-03-01:
The DNS redirect sends updates to your Acuparse server. Acuparse takes the update and rips the data, then sends the update off to MyAcuRite. Same as if your hub/access uploaded it directly. You can enable/disable the uploading of data in the same way as WU, PWS, or CWOP in the config. By default, Acuparse will send the update data to MyAcuRite.

bdwilson, 2018-03-01:
Awesome!  Looking for alternatives before the EOL date. Appreciate your work on this!

closed 2018-03-01 by bdwilson