acuparse/acuparse #27: Problems with database creation - Ubuntu

maxwellpower, 2018-03-06:
Continuing from #20

bp2008, 2018-03-06:
```
brian@acuparse:~$ mysql -uroot -p${ROOTPW} -e "DELETE FROM mysql.user WHERE User='';DELETE FROM mysql.user WHERE User='root' AND Host NOT IN ('localhost', '127.0.0.1', '::1');DROP DATABASE IF EXISTS test;DELETE FROM mysql.db WHERE Db='test' OR Db='test\\_%';FLUSH PRIVILEGES;"
Enter password:
brian@acuparse:~$ mysql -uroot -p${ROOTPW} -e "CREATE DATABASE acuparse; GRANT ALL PRIVILEGES ON acuparse.* TO acuparse@localhost IDENTIFIED BY '$DBPW'; GRANT SUPER, EVENT ON *.* TO acuparse@localhost"
Enter password:
ERROR 1133 (42000) at line 1: Can't find any matching row in the user table
brian@acuparse:~$
```

maxwellpower, 2018-03-06:
Replace ${ROOTPW} with your root password and $DBPW with the password you want for the Acuparse database.

bp2008, 2018-03-06:
Alright, I had to delete and recreate the acuparse database.  But then it worked.

```
brian@acuparse:~$ mysql -uroot -pROOTPASS -e "CREATE DATABASE acuparse; GRANT ALL PRIVILEGES ON acuparse.* TO acuparse@localhost IDENTIFIED BY 'USERPASS'; GRANT SUPER, EVENT ON *.* TO acuparse@localhost"
mysql: [Warning] Using a password on the command line interface can be insecure.
ERROR 1007 (HY000) at line 1: Can't create database 'acuparse'; database exists
brian@acuparse:~$ mysql -uroot -pROOTPASS -e "DROP DATABASE IF EXISTS acuparse;DELETE FROM mysql.db WHERE Db='acuparse' OR Db='acuparse\\_%';FLUSH PRIVILEGES;"
mysql: [Warning] Using a password on the command line interface can be insecure.
brian@acuparse:~$ mysql -uroot -pROOTPASS -e "CREATE DATABASE acuparse; GRANT ALL PRIVILEGES ON acuparse.* TO acuparse@localhost IDENTIFIED BY 'USERPASS'; GRANT SUPER, EVENT ON *.* TO acuparse@localhost"
mysql: [Warning] Using a password on the command line interface can be insecure.
brian@acuparse:~$
```

After this, I was able to proceed beyond the database configuration page, create a user, and now I am on the Modify System Settings page.

Of course I have no idea why the installer had trouble with this.  This was a fresh install of Ubuntu Server 16.04.4 LTS downloaded today from https://www.ubuntu.com/download/server and running on an ESXi 6.5 box.

bp2008, 2018-03-06:
Thank you for your assistance.

commit 054290a8 (2018-07-07): documentation updates

fix feels like and update text

closes #27
commit ed47ade0 (2018-07-07): documentation updates

fix feels like and update text

closes #27
commit 441a7fea (2018-07-07): documentation updates

fix feels like and update text

closes #27
closed 2018-07-08 by maxwellpower