acuparse/acuparse #28: Room Monitor with Water Detector = (ACCESS)[ERROR]: Unknown Sensor

bp2008, 2018-03-06:
I have three of these: https://www.acurite.com/room-monitor-with-water-detector-00276wd.html

Acuparse doesnt' recognize them.

```
Mar  6 14:53:20 acuparse acuparse(2.2.2): (ACCESS)[ERROR]: Unknown Sensor 00007540. Raw = dateutc=2018-03-06T21:53:01&id=24C86EXXXXXX&mt=ProIn&sensor=00007540&sensorbattery=normal&rssi=3&hubbattery=normal&baromin=30.35&indoorhumidity=36&indoortempf=68.3&probe=1&check=0&water=0&dewptf=40.3
Mar  6 14:53:39 acuparse acuparse(2.2.2): (ACCESS)[ERROR]: Unknown Sensor 00007488. Raw = dateutc=2018-03-06T21:53:01&id=24C86EXXXXXX&mt=ProIn&sensor=00007488&sensorbattery=normal&rssi=4&hubbattery=normal&baromin=30.35&indoorhumidity=34&indoortempf=69.0&probe=1&check=0&water=0&dewptf=39.5
Mar  6 14:53:39 acuparse acuparse(2.2.2): (ACCESS)[ERROR]: Unknown Sensor 00007821. Raw = dateutc=2018-03-06T21:53:01&id=24C86EXXXXXX&mt=ProIn&sensor=00007821&sensorbattery=normal&rssi=0&hubbattery=normal&baromin=30.35&indoorhumidity=65&indoortempf=41.1&probe=1&check=0&water=0&dewptf=30.2
```

bp2008, 2018-03-06:
It seems the data from these sensors is not forwarded on to acurite's cloud either.

maxwellpower, 2018-03-07:
Will work on adding into the next version. Right now, it only supports a 5N1 and Towers. I have to make a bunch of changes to support the Atlas and multiple sensors.

I originally just wrote it for myself and I didn't have any other sensors to play with. Will slowly get them all added as others send strings.

bp2008, 2018-03-07:
Yup, understandable.  Would be so nice if AcuRite didn't make us go to so much work to get access to our own data.  It would be so easy for them to just serve the weather data from the device's embedded web server.

maxwellpower, 2018-03-07:
It's true. It's really an interesting situation. Since they are providing this "cloud" service for free, forever just because you buy the hardware. I have never been able to figure out the revenue model there. Since the cost to run their infrastructure must be astronomical. So I can only assume we are paying with our data, the same as any other free cloud service. How they are monetising our data is still a good guess.

Getting the data directly from the Access/Hub would be cool. I have thought a bunch about it and don't think it would really help. Since we would then have to pull the data from the device for archiving still. 

This way, the device is dumping it to us without having to ask. I don't believe there is a better way just yet, which is why I think AcuRite has the devices sending rather than just receiving and displaying. If they were able to make it work without MyAcuRite, I think they would have and avoided the cloud compute cost.

Also, for NAT reasons. It would be really challenging to educate some users on how to open their firewall etc to get access to their readings.

At the moment the Access device technically lets us set the server name it uploads too. They made it a lot easier for us to get our data out. If they would just change the update interval on the Access to update every minute, that would be awesome.

bp2008, 2018-03-07:
I have to admit there are situations where having the device push the data is more convenient than pulling it, and the new Access device is a lot better for this than the old SmartHUBs since we don't have to muck around with DNS and DHCP servers anymore.  Of course they could lock that down really easily if they wanted to just by hard coding the host name and validating the server certificate.  I hope they never do.

I understand why they would reduce the update interval.  It lowers their server costs, I am sure.  Now the data seems to go out in bursts of multiple http requests using the same TCP connection.  Something related that I noticed is that normally their service responds with the time zone, e.g. `{"timezone":"-07:00"}` but sometimes it responds with `{}` and this apparently causes the Access device to abort the rest of its http requests for that particular data burst.  I wonder if this is a bug or if it is some kind of throttling mechanism on their end.

maxwellpower, 2018-03-07:
They could lock it down. I think having to add their cert to the Access would be a no go. This would require new certs to come out in firmware and some won't get the new firmware. Would be nothing but problems for them. They could lock out the server change in the firmware but I don't see why. They added the ability to change it for a reason one would think. It had to be coded. I am hoping they embrace us, the more advanced users in their community, rather than shun. Maybe some API documentation or something would be great.

MyAcuRite does respond with the timezone but it seems only when a sensor is added to MyAcuRite. If it is not added, MyAcuRite responds with `{}`. I have not noticed the `{}` causing update issues personally. Right now for testing only my 5N1 is added to MyAcuRite, the towers are not. Still seems to update Acuparse OK.

I did notice some weird issues with Ubuntu while doing my testing but have not been able to isolate it. Seems to only happen with the Access, but Ubuntu will drop or lose the updates. The same does not happen on Debian. Has been interesting to track down. Since I'm not even sure if it's widespread, or just me.

commit 268f5573 (2018-04-03): Support for Indoor/Outdoor Monitors

Resolves #28 and Resolves #30
commit 2761b6c8 (2018-04-03): Support for Indoor/Outdoor Monitors

Resolves #28 and Resolves #30
closed 2018-04-04 by maxwellpower