acuparse/acuparse #39: Cannot install Acuparse

olddeadhacker, 2018-07-16:
Tying to install Acuparse on Raspberry Pi 3 B+. Tried this on 07/15/18 in the PM MST. Followed Mike Frazier's script. All went well until end. Cmd to install Acuparse showed multiple errors. Did not have logging enabled for 1st pass, but noticed many adds and removes for apache2. Have putty.log for 3rd try. Errors say missing many .php2 modules missing. Example here.... "Package libapache2-mod-php7.2 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or is only available from another source"

Also.... "E: Package 'libapache2-mod-php7.2' has no installation candidate". Many of these to match all missing modules in first set.

Apache seems to be a basket case.  Still get "Acuparse Installation Complete!" however.

Can supply complete log if needed. Do not know if I was attempting install while your maintenance was being done on 07/15/18. Would like to start Acuparse install over completely without having to install op system. Have already deleted install.sh, install.sh.1, and install.sh.2. Need to delete many more files I assume but not sure which ones.  

Have no /opt/acuparse/src/usr. Obviously have no web page. Can access Pi via puTTY.

Ideas? Could be disconnect between brain and keyboard here. Thanks for your help.

maxwellpower, 2018-07-16:
I don't really support PI installs directly. Only Ubuntu/Debian. Pi support is provided through community members.

That being said:
Sounds like an issue instaling PHP7.2 based on your notes. It's possible the script does not detect the PI properly or that the Sury PHP repo does not support the PI yet.

Try the following, in order:
1) Try installing PHP7.2 Manually	
	
`apt-get install ca-certificates apt-transport-https -y`

`wget -q https://packages.sury.org/php/apt.gpg -O- | apt-key add -`

`echo "deb https://packages.sury.org/php/ stretch main" | tee /etc/apt/sources.list.d/php.list`

`apt-get update`

`apt-get install git ntp imagemagick exim4 apache2 mysql-server php7.2 libapache2-mod-php7.2 php7.2-mysql php7.2-gd php7.2-curl php7.2-json php7.2-cli php7.2-common -y`

2) If that does not work then try PHP7.0 instead:

`rm /etc/apt/sources.list.d/php.list`

`apt-get update && apt-get install git ntp imagemagick exim4 apache2 mysql-server php7.0 libapache2-mod-php7.0 php7.0-mysql php7.0-gd php7.0-curl php7.0-json php7.0-cli php7.0-common -y`

olddeadhacker, 2018-07-16:
Thanks for the prompt reply. Understand you yourself do not support Acuparse on a Pi. I used Mike Frazier's write-up found on the Acuparse site, including the last items to install Acuparse.  Last conversations with Mike over on Acurite Community Forum a few days ago, he mentioned that his instructions worked for him. 

Will try your suggestions, in recommended order, with logging on, and report back. May be a few days. Thanks for the advice.

olddeadhacker, 2018-07-20:
Started over completely. Reloaded OS. Reloaded Acuparse. Now able to access homepage. Error probably was cockpit error here.

On homepage entered pw for Acuparse DB, clicked Save, no response, connection spun, clicked SignIn, got a default page, and msg 'live data temporarily unavailable'. Not sure what UID to use where or when. Will continue setup process.

mfrazzz, 2018-07-20:
> After the install finishes, you will be redirected to the homepage. You
> then enter your database information. It will then ask you to create the
> first user. Only after the user is created is set up actually complete.
>
> If you're having issues with that part, you can remove your config file
> and try again. If there is no user in the database, you're going to have
> issues.
>
> rm /opt/acuparse/src/usr/config.php
>
> To find the sensor ID's the easiest way is to check your syslog for the
> errors.
>
> tail -f /var/log/syslog

If you haven't figured it out before hand, I'll take a look this weekend
(have stuff going on today, and tonight so doubt I can look).  I have not
specifically done Apache 7.2 yet (I am running the latest version on my Pi
3).  Not sure Apache 7.2 is in the Debian/Raspbian Stretch repo (I've done
upgrades and don't remember that upgrading, but I could be wrong).

As Max mentioned, if the install went fine, but its just the web page where
there is a problem, it seems if there is an issue connecting to the
database the first time then the step to create the user doesn't happen and
then the install isn't complete.  Just delete the config.php file and then
open the web page again.

olddeadhacker, 2018-07-20:
Already did 3 iterations of removing config file before I posted the above 4 hours previous to this post. After 4th iteration of removing the config file, am up OK. Made first user, can log in and out to that UID. I did change the browser that I used to access the homepage on the 4th try to enter DB info., from Pale Moon to IE11 (That's going backwards somewhat!), so that's another variable.

Got hung up a few minutes at entering lat. and lon. I entered 6 digits to right of decimal point; didn't realize Acuparse only accepted 3.  Have sensor IDs from smartHub when it is plugged directly into router; will finish that portion up later.

Went back in with Pale Moon and all seems good.

Thanks Maxwell and Mike; appreciate the help.

olddeadhacker, 2018-07-22:
I believe last of my problems is fixed. Problem was very slow response from Acuparse. Fix = change data forward option to Acurite to 'alternate'. Will leave data going to Acurite that way for a while. Thanks again for all the help.

closed 2018-07-22 by olddeadhacker