acuparse/acuparse #40: Observations/Questions regarding dsiplayed data on Acuparse home page

olddeadhacker, 2018-07-24:
Sir,

Some questions/observations regarding the displayed data on my Acuparse home page.

Installations particulars here as follows...

Acuparse installed on a Raspberry Pi 3 B+ on 07/19/2018. Raspbian Stretch Lite as of 07/03/2018, PHP 7.2, Apache2 2.4.25-3, Sqlite 1.5.4-3.

Pi up continuously now for > 56 hours. Upload only to MyAcurite.com using secondary. I have attached a screen shot (some icons did not xfer when page was saved - not a problem - all are OK on display).

Temperature, Humidity, Wind, Rain from the 5n1 agree with what is shown on my account at MyAcuRite.com and on my indoor console, +/- a degree or so. Same with T and H from two T/H sensors.

Observations/Questions/Problems...

1) The Low and High temperatures shown under the 5n1 section are incorrect both to temperature itself and the time. At first, one or two days ago, the time seemed to be set to UTC, ignoring my setting, but as of 7/24/18 I do not know. The temperature and time now seem to be frozen. What is shown makes no sense. I looked at the same data on your web page and Mike Frazier's web page and they make sense; seem to be accurate. Does Acuparse calculate this?

2) If I change the Date and Time Display shown in the upper center, as can be seen on the screen shot, I expected the format used under 'Moon' to follow that for consistency. It did not. Is the format under 'Moon' supposed to change?

Thanks for your comments.

[Live Weather _ Weather Station _ Catalina, Arizona, USA.pdf](https://github.com/acuparse/acuparse/files/2224709/Live.Weather._.Weather.Station._.Catalina.Arizona.USA.pdf)

maxwellpower, 2018-07-24:
Just a quick guess without investigating deeper:
You are using Sqlite while the script was designed for MySQL. I would expect there might be some incompatibilities.

Times are calculated using the database and Acuparse. Depending on what is being performed. It's critical that your system timezone and the timezone in your Acuparse config are the same. MySQL will use the system time. Calculations like moon time use your Acuparse config. The time set in Acuparse ultimately only affects the PHP timezone setting. I would check and see what Sqlite is using for the time.

Setting the time format in the config is really only for the header display. A lot of the time values are hardcoded for displayability. Since it would be hard to use the full format for every reading. So for moon time, it's hardcoded to display as `j M @ H:i`.

I can probably add a drop-down selector at some point to allow for the quick modification of the other display formats.

olddeadhacker, 2018-07-24:
Thanks for your quick reply. I misspoke in my original post above. I am using MySQL. See attached output of 'dpkg --install'. Earlier today I just read thru my install session that I had logged thru puTTY and went off on a tangent!

My system timezone and the Acuparse timezone setting is both MST (in Windows, it's Arizona (UTC-7.00; in Acuparse = America, Phoenix). I get the same result using two different PCs.

I brought up my web page a few minutes ago. The High and Low temperature and times have changed. The low temp is now correct (compared against the data on MyAcuRite.com) but the time is UTC. Low temp ~~ 76 degrees F really at 04:30 AM, Acuparse says 76.3 at11:28. 

Acuparse says right now High temp = 108 F at 21:07. Display s/b 108 at 14:10 (so says MyAcuRite.com). So the 21:07 says UTC.!  So now High temp = OK, but time here also UTC. My time right now as I write this = 24/07/18 at 14:33 MST +/-.

[putty 2.log](https://github.com/acuparse/acuparse/files/2225529/putty.2.log)

maxwellpower, 2018-07-24:
UTC is +6 Mountain at the moment. You might have duplicated readings stored for both UTC and MST until you catch up to what UTC was when you made the adjustment. Clearing the database tables can sometimes solve this. MySQL might even notice the change and make the adjustment for you.

Based on what you noted, it seems like duplicates might be skewing your data but also notice you mention your Windows time zone. Note that we're talking about your Acuparse server and not the workstation you are using to view the dashboard. The timezone should be set exactly the same in the OS and Acuparse config. `America/Phoenix`.

To be sure, run the following on your Acuparse server **AS ROOT**:
`dpkg-reconfigure tzdata && systemctl restart rsyslog.service && systemctl restart mysql.service`

Also of note, once you change the time zone you must either reboot or restart the Syslog and MySQL services. This is done for you in that snippet above.

Will update the install doc and the installer to deal with the time zone. It's something I definitely overlooked while writing the install process. I should definitely not assume the OS timezone is set before installing.

Forgot to mention earlier, you're missing icons because FontAwesome does not appear to be loading. Make sure nothing is blocking this code from running `<script defer src="/lib/mit/fontawesome/js/all.js"></script>`. Suspect you might not have Javascript support enabled/installed. Acuparse uses a bit of Javascript. For best results, test using Chrome or Firefox. Safari, Opera, and Edge should also work just as well; anything else is a crapshoot.

mfrazzz, 2018-07-25:
> Yes, I made another mistake. I referred to Windows when I should have
> referred to Raspbian on the Pi. I rechecked the time set on the Pi and it
> was not set to America-Arizona. I thought I already did, so I messed that
> up. It Is now set correctly. I rebooted the system and the time problem
> seems fixed.
>
> Did all this before I read your latest comments.
>
> Actually Mtn Standard Time is now the same as Pacific Daylight Time
> because AZ (except the Navajo Nation) does not observe DST , so we are same
> as UTC-08:00 until November sometime. That is why AZ (or Phoenix) has a
> separate listing. Same with Hawaii.
>
> Good point on adding a reminder to check that the time zone the OS is
> using is correct and agrees Acuparse.
>
> All the icons are fine when my page is up. The missing ones happened on
> the output of the page when I saved the page. Yes, I do have some
> Javascript blocked for security purposes; I will remember your advice on
> this.
>
> Thanks very much for your time and patience. Where is the best place for
> coffee up you way? Tim Hortons? I owe you 3 or 4 cups!

I think I'll update my guide to set the timezone on Raspbian before the
Acuparse install happens.

Is there a way for Acuparse to see the OS timezone setting and warm if it's
different than what is set in the app?

olddeadhacker, 2018-07-25:
Mike,

Do you think it would be worthwhile to mention in your writeup that a Raspberry Pi does not contain a real time clock as 'standard equipment'? (For those of us with minimal experience with this hardware.) It's inexpensive to add one. I may do that as BU. Anyway FWIW.

All is well on my system now.

maxwellpower, 2018-07-25:
I'm glad you got it working and thanks for the education on the timezone. I had no idea. Been to Arizona a few times but never took the time to look into the timezone deeper. I am also MST but in Alberta.

I graciously accept all the coffee :) Tim's has gone downhill these days, McD's is where it's at today.

Mike,
I don't know of a way to check. I did some quick searching around, and PHP is pretty finicky about timezones, that's why we set it in Acuparse, to be sure. The problem is MySQL because it uses the system time by default. It's not super easy to change that other than in its config. Plus, the Syslog needs to be correct too.

So, in the installer, I added:
` dpkg-reconfigure tzdata `
` systemctl restart rsyslog.service`
At the end of the initial questions. So we force the new user to choose a system timezone; at the very least. I added a blurb about it into the install guide to run those commands as well.

closed 2018-07-31 by maxwellpower