acuparse/acuparse #45: Work with PHP 5.x?

kevinkey619, 2018-10-31:
Is there a way to make this work with PHP 5.x? PHP 7.2 absolutely refuses to install on my Raspberry Pi 3.

maxwellpower, 2018-10-31:
It should work just fine on 5.x but I no longer pay any attention to it. So things could break. It was previously supported using 5 before moving to 7.0 and then 7.2 not so long ago.

Try using 7.0 instead of 5.x for better support:
`apt-get update && apt-get install git ntp imagemagick exim4 apache2 mysql-server php7.0 libapache2-mod-php7.0 php7.0-mysql php7.0-gd php7.0-curl php7.0-json php7.0-cli -y`

If you must use 5.x:
`apt-get update && apt-get install git ntp imagemagick exim4 apache2 mysql-server php5 libapache2-mod-php5 php5-mysql php5-gd php5-curl php5-json php5-cli -y`

Interesting why you can't get 7.2 on PI. I don't have any and haven't played with it much but I know there are users with PI's. This is the first I've heard of issues with PHP. Will have to investigate some more.

kevinkey619, 2018-10-31:
Hi Maxwell,

Thanks, but still no luck. Also, ever time I run the installer it takes
nearly an hour and seems like it's totally reinstalling everything on my
Raspberry Pi - including the OS and all.

kevinkey619, 2018-10-31:
This seems to not be designed for the Raspberry Pi. Besides reinstalling every package known to man every time I run install.sh, now I get the below errors:

E: Could not open lock file /var/lib/dpkg/lock-frontend - open (13: Permission denied)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), are you root?

kevinkey619, 2018-10-31:
I'm running the install yet again - and having rebooted yet again - and this is just a small sampling of what it's doing (and will do for the next hour or so) Is the entire OS being reinstallled?

Get:269 http://raspbian.phirephly.design/raspbian buster/main armhf moreutils armhf 0.62-1 [68.6 kB]
Get:270 http://raspbian.phirephly.design/raspbian buster/main armhf ncdu armhf 1.13-1 [37.6 kB]
Get:271 http://raspbian.phirephly.design/raspbian buster/main armhf openresolv armhf 3.8.0-1 [22.9 kB]
Get:272 http://raspbian.phirephly.design/raspbian buster/main armhf p11-kit armhf 0.23.14-2 [248 kB]
Get:273 http://raspbian.phirephly.design/raspbian buster/main armhf pidgin-otr armhf 4.0.2-2 [116 kB]
Get:274 http://raspbian.phirephly.design/raspbian buster/main armhf pinentry-gtk2 armhf 1.1.0-1 [64.1 kB]
Get:275 http://raspbian.phirephly.design/raspbian buster/main armhf po-debconf all 1.0.20 [247 kB]
Get:276 http://raspbian.phirephly.design/raspbian buster/main armhf poppler-utils armhf 0.63.0-2 [149 kB]
Get:277 http://raspbian.phirephly.design/raspbian buster/main armhf powermgmt-base all 1.33 [11.0 kB]
Get:278 http://raspbian.phirephly.design/raspbian buster/main armhf ppp armhf 2.4.7-2+3 [313 kB]
Get:279 http://raspbian.phirephly.design/raspbian buster/main armhf python-apt-common all 1.7.0 [94.5 kB]
Get:280 http://raspbian.phirephly.design/raspbian buster/main armhf python-blinker all 1.4+dfsg1-0.2 [13.9 kB]
Get:281 http://raspbian.phirephly.design/raspbian buster/main armhf python-bluez armhf 0.22+really0.22-1 [52.3 kB]
Get:282 http://raspbian.phirephly.design/raspbian buster/main armhf python-setuptools all 40.4.3-1 [375 kB]
Get:283 http://raspbian.phirephly.design/raspbian buster/main armhf python-pkg-resources all 40.4.3-1 [180 kB]
Get:284 http://raspbian.phirephly.design/raspbian buster/main armhf python-chardet all 3.0.4-1 [80.2 kB]
Get:285 http://raspbian.phirephly.design/raspbian buster/main armhf python-colorama all 0.3.7-1 [25.7 kB]
Get:286 http://raspbian.phirephly.design/raspbian buster/main armhf python-crypto armhf 2.6.1-9 [252 kB]
Get:287 http://raspbian.phirephly.design/raspbian buster/main armhf python-distlib all 0.2.8-1 [122 kB]
Get:288 http://raspbian.phirephly.design/raspbian buster/main armhf python-dnspython all 1.15.0-1 [102 kB]
Get:289 http://raspbian.phirephly.design/raspbian buster/main armhf python-gobject-2 armhf 2.28.6-13 [258 kB]
Get:290 http://raspbian.phirephly.design/raspbian buster/main armhf python-gtk2 armhf 2.24.0-5.1 [499 kB]
Get:291 http://raspbian.phirephly.design/raspbian buster/main armhf python-itsdangerous all 0.24+dfsg1-2 [13.0 kB]
Get:292 http://raspbian.phirephly.design/raspbian buster/main armhf python-markupsafe armhf 1.0-1+b1 [13.7 kB]
Get:293 http://raspbian.phirephly.design/raspbian buster/main armhf python-jinja2 all 2.10-1 [106 kB]
Get:294 http://raspbian.phirephly.design/raspbian buster/main armhf python-pyasn1 all 0.4.2-3 [57.6 kB]
Get:295 http://raspbian.phirephly.design/raspbian buster/main armhf python-ndg-httpsclient all 0.5.1-1 [25.0 kB]
Get:296 http://raspbian.phirephly.design/raspbian buster/main armhf python-pip all 9.0.1-2.3 [179 kB]
Get:297 http://raspbian.phirephly.design/raspbian buster/main armhf python-pyinotify all 0.9.6-1 [26.8 kB]
Get:298 http://raspbian.phirephly.design/raspbian buster/main armhf python-rpi.gpio armhf 0.6.3-1+b1 [20.1 kB]
Get:299 http://raspbian.phirephly.design/raspbian buster/main armhf python-serial all 3.4-4 [83.0 kB]
Get:300 http://raspbian.phirephly.design/raspbian buster/main armhf python-simplejson armhf 3.15.0-1 [69.7 kB]
Get:301 http://raspbian.phirephly.design/raspbian buster/main armhf python-six all 1.11.0-2 [15.1 kB]
Get:302 http://raspbian.phirephly.design/raspbian buster/main armhf python-werkzeug all 0.14.1+dfsg1-3 [196 kB]
Get:303 http://raspbian.phirephly.design/raspbian buster/main armhf python3-colorama all 0.3.7-1 [18.1 kB]
Get:304 http://raspbian.phirephly.design/raspbian buster/main armhf python3-distlib all 0.2.8-1 [122 kB]
Get:305 http://raspbian.phirephly.design/raspbian buster/main armhf python3-jinja2 all 2.10-1 [106 kB]
Get:306 http://raspbian.phirephly.design/raspbian buster/main armhf python3-pip all 9.0.1-2.3 [142 kB]
Get:307 http://raspbian.phirephly.design/raspbian buster/main armhf python3-pyinotify all 0.9.6-1 [26.9 kB]
Get:308 http://raspbian.phirephly.design/raspbian buster/main armhf python3-urllib3 all 1.24-1 [96.8 kB]
Get:309 http://raspbian.phirephly.design/raspbian buster/main armhf python3-werkzeug all 0.14.1+dfsg1-3 [196 kB]
Get:310 http://raspbian.phirephly.design/raspbian buster/main armhf qtcore4-l10n all 4:4.8.7+dfsg-17+rpi1 [661 kB]
Get:311 http://raspbian.phirephly.design/raspbian buster/main armhf rfkill armhf 2.32.1-0.1 [81.6 kB]
Get:312 http://raspbian.phirephly.design/raspbian buster/main armhf rsync armhf 3.1.2-2.2 [355 kB]
Get:313 http://raspbian.phirephly.design/raspbian buster/main armhf rtkit armhf 0.11-6 [32.6 kB]
Get:314 http://raspbian.phirephly.design/raspbian buster/main armhf ruby-full all 1:2.5.1 [10.0 kB]
Get:315 http://raspbian.phirephly.design/raspbian buster/main armhf rubygems-integration all 1.11 [4,994 B]
Get:316 http://raspbian.phirephly.design/raspbian buster/main armhf scrot armhf 0.8-18 [18.3 kB]
Get:317 http://raspbian.phirephly.design/raspbian buster/main armhf squeak-plugins-scratch armhf 1.4.0.2~svn.r83-2.1 [24.9 kB]
Get:318 http://raspbian.phirephly.design/raspbian buster/main armhf squeak-vm armhf 1:4.10.2.2614-8 [430 kB]
[... cut ...] ttp://raspbian.phirephly.design/raspbian buster/main armhf strace armhf 4.21-1 [202 kB]
Get:320 http://raspbian.phirephly.design/raspbian buster/main armhf sudo armhf 1.8.23-2 [1,158 kB]
Get:321 http://raspbian.phirephly.design/raspbian buster/main armhf python-meld3 all 1.0.2-2 [37.3 kB]
Get:322 http://raspbian.phirephly.design/raspbian buster/main armhf supervisor all 3.3.1-1.1 [280 kB]
Get:323 http://raspbian.phirephly.design/raspbian buster/main armhf tango-icon-theme all 0.8.90-7 [1,254 kB]

kevinkey619, 2018-10-31:
And it's still installing. 

Removing obsolete conffile /etc/bash_completion.d/pon ...
Setting up pidgin-otr (4.0.2-2) ...
Setting up libltdl-dev:armhf (2.4.6-6) ...
Setting up libvisual-0.4-plugins:armhf (1:0.4.0+dfsg1-11) ...
Setting up libhttp-cookies-perl (6.04-1) ...
Setting up xinit (1.4.0-1) ...
Setting up portaudio19-dev:armhf (19.6.0-1) ...
Setting up libv4l-0:armhf (1.14.2-1) ...
Setting up libmenu-cache3:armhf (1.1.0-1) ...
Setting up python-jinja2 (2.10-1) ...
Setting up libpolkit-agent-1-0:armhf (0.105-21) ...
Setting up python-gtk2 (2.24.0-5.1) ...
Setting up testng (6.9.12-4) ...
Setting up libhcrypto4-heimdal:armhf (7.5.0+dfsg-2) ...
Setting up v4l-utils (1.14.2-1) ...
Setting up moreutils (0.62-1) ...
Setting up tcl (8.6.0+9) ...
Setting up libsdl-mixer1.2:armhf (1.2.12-15) ...
Setting up x11-utils (7.7+4) ...
Installing new version of config file /etc/X11/app-defaults/Editres ...
Setting up libnss3:armhf (2:3.39-1) ...
Setting up libpcap0.8-dev:armhf (1.8.1-6) ...
Setting up tightvncserver (1:1.3.9-9+b5) ...
Setting up libxvmc1:armhf (2:1.0.10-1) ...
Setting up libinput-bin (1.12.1-1) ...
Setting up tcpdump (4.9.2-3) ...
Setting up libmbim-proxy (1.16.0-1) ...
Setting up libjackrabbit-java (2.14.6-1) ...
Setting up libsoftware-license-perl (0.103013-2) ...
Setting up haveged (1.9.1-6) ...
Setting up libtevent0:armhf (0.9.37-1) ...
Setting up xine-ui (0.99.9-1.3+b1) ...
Updated the MIME types in xine's menu file.
Setting up libwacom-bin (0.31-1) ...
Setting up libtk8.5:armhf (8.5.19-3) ...
Setting up libpolkit-backend-1-0:armhf (0.105-21) ...
Setting up tk (8.6.0+9) ...
Setting up libnunit-console-runner2.6.3-cil (2.6.4+dfsg-1) ...
* Installing 1 assembly from l

kevinkey619, 2018-10-31:
And finally... after all that...

php7.2-json is already the newest version (7.2.11-2+0~20181015120859.6+stretch~1.gbp8105e0).
php7.2-json set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 php7.2-curl : Depends: libcurl3 (>= 7.44.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
END: Packages

Install Acuparse from git
Reinitialized existing Git repository in /opt/acuparse/.git/
fatal: remote origin already exists.
Already on 'master'
Your branch is up-to-date with 'origin/master'.
END: Acuparse

Configuring Apache
rm: cannot remove '/etc/apache2/sites-available/000-default.conf': No such file or directory
rm: cannot remove '/etc/apache2/sites-available/default-ssl.conf': No such file or directory

kevinkey619, 2018-10-31:
And now its removing a bunch of stuff?

((Reading database ... 192544 files and directories currently installed.)
Removing libgeronimo-jpa-2.0-spec-java (1.1-3) ...
Removing aspectj (1.9.2~rc2-1) ...
Removing coinor-libcoinmp1:armhf (1.7.6+dfsg1-1) ...
Removing coinor-libcbc3 (2.8.12-1) ...
Removing coinor-libcgl1 (0.58.9-1) ...
Removing coinor-libclp1 (1.15.10-1) ...
Removing coinor-libosi1 (0.106.9-1) ...
Removing coinor-libcoinutils3 (2.9.15-3) ...
Removing libesd0:armhf (0.2.41-11) ...
Removing esound-common (0.2.41-11) ...
Removing fop (1:1.1.dfsg2-1+deb8u1) ...
Removing gnupg-agent (2.2.10-3+rpi1) ...
Removing idle-python2.7 (2.7.15-4) ...
Removing libabw-0.1-1:armhf (0.1.2-1) ...
Removing libgeronimo-osgi-support-java (1.1-1) ...
Removing libaspectj-java (1.9.2~rc2-1) ...
Removing libaudiofile1:armhf (0.3.6-4) ...
Removing liborcus-0.8-0 (0.7.0+dfsg-9) ...
Removing libboost-system1.55.0:armhf (1.55.0+dfsg-3) ...
Removing libcdr-0.1-1:armhf (0.1.4-1+b1) ...
Removing lp-solve (5.5.0.13-7+b1) ...
Removing libcolamd2.8.0:armhf (1:4.2.1-3) ...
Removing libe-book-0.1-1:armhf (0.1.3-1+b1) ...
Removing libelfg0:armhf (0.8.13-5) ...
Removing libetonyek-0.1-1:armhf (0.1.8-2) ...
Removing libfile-copy-recursive-perl (0.44-1) ...
Removing libfreehand-0.1-1 (0.1.2-2) ...
Removing libgcr-3-common (3.14.0-2) ...
Removing libgltf-0.0-0 (0.0.2-2) ...
Removing liblist-moreutils-perl (0.33-2+b1) ...
Removing libmono-corlib4.0-cil (5.10.1.20-0xamarin1+debian7b1) ...
Removing libmspub-0.1-1:armhf (0.1.4-1+b1) ...

kevinkey619, 2018-10-31:
Do I need to format the drive on my Raspberry Pi every time I want to install this? Reinstalling all my other apps, etc. will be a pain.

kevinkey619, 2018-10-31:
I've uninstalled and reinstalled numerous times. This is what I see when I go to my Raspberry Pi in my browser. 

?php
/**
 * Acuparse - AcuRiteÂ®â€Ž Access/smartHUB and IP Camera Data Processing, Display, and Upload.
 * @copyright Copyright (C) 2015-2018 Maxwell Power
 * @author Maxwell Power <max@acuparse.com>
 * @link http://www.acuparse.com
 * @license AGPL-3.0+
 *
 * This code is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Affero General Public License as published
 * by the Free Software Foundation, either version 3 of the License, or
 * any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU Affero General Public License for more details.
 *
 * You should have received a copy of the GNU Affero General Public License
 * along with this code. If not, s

kevinkey619, 2018-11-01:
I finally got it to work! I decided to format the hard drive (SD card) on my Raspberry Pi 3, installed the latest Raspbian, and start from scratch. I downloaded and ran the install script and it worked flawlessly.

closed 2018-11-01 by kevinkey619