acuparse/acuparse #46: Can't get past homepage

kevinkey619, 2018-11-01:
I finally got this to install, now I can't get past the homepage. I specified a username and password. Now every time I go to the homepage it just displays the number "1".  What am I missing?

mfrazzz, 2018-11-01:
You can delete the config file under /opt/acuparse directory tree and then
pull up the website, it will have you setup the username again which
usually fixes that issue

kevinkey619, 2018-11-01:
Oh, I've already done that that a thousand times. Same result every time.
Where in the instructions am I missing the steps to set this thing up via
the homepage? What am I doing wrong?

kevinkey619, 2018-11-01:
Are there any instructions anywhere? What am I doing wrong?!

kevinkey619, 2018-11-01:
I deleted the config file and tried to specify an admin name and password yet again. Still just seeing a 1 on the screen. Are there any instructions for this?

kevinkey619, 2018-11-01:
Still having the same problem.

closed 2018-11-06 by maxwellpower

csmithsound, 2019-03-07:
I spent hours trying to get bast the homepage,
After recreating config file i needed to delete the database and recreate following thread 27.

MoonFri, 2019-04-17:
Start with the follwoing commend to start fresh. 
rm /opt/acuparse/src/usr/config.php

Then pull up the website and click on the X to close the green box (don't recall what it states) Then enter your information. Worked for me after many many attempts and 2 reloads. Seems the alert box must be closed first. Hope it helps.