acuparse/acuparse #48: Worked very briefly, now it's no longer receiving weather data

kevinkey619, 2018-11-01:
This worked briefly, now it's stopped receiving weather data from my Access. Is this compatible with Raspberry Pi 3? Do I have to wipe everything and install yet again?

kevinkey619, 2018-11-01:
How do I configure this to receive sensor data updates from my Access more frequently? It's been over an hour and that data hasn't refreshed.

maxwellpower, 2018-11-01:
Although the script should work on rasberry PI, I don't directly support it. Only an actual Debian/Ubuntu install. I have never installed or tested this on a PI myself. There is some documentation in the wiki users have provided to assist other PI users.

The script is only going to receive the updates it's sent. Nothing more, nothing less.

The Access sends out updates every five minutes. If they are being received, you will see them appear in the syslog. `tail -f /var/log/syslog`

closed 2018-11-01 by maxwellpower