acuparse/acuparse #49: Install failing

kevinkey619, 2018-11-01:
nabling conf serve-cgi-bin.
Enabling site 000-default.
Created symlink /etc/systemd/system/multi-user.target.wants/apache2.service → /lib/systemd/system/apache2.service.
Created symlink /etc/systemd/system/multi-user.target.wants/apache-htcacheclean.service → /lib/systemd/system/apache-htcacheclean.service.
Job for apache2.service failed because the control process exited with error code.
See "systemctl status apache2.service" and "journalctl -xe" for details.
invoke-rc.d: initscript apache2, action "start" failed.
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-10-31 23:05:27 PDT; 33ms ago
  Process: 2654 ExecStart=/usr/sbin/apachectl start (code=exited, status=127)

Oct 31 23:05:27 raspberrypi3 systemd[1]: Starting The Apache HTTP Server...
Oct 31 23:05:27 raspberrypi3 apachectl[2654]: /usr/sbin/apachectl: 174: /usr/sbin/apachectl: /usr/sbin/ap… found
Oct 31 23:05:27 raspberrypi3 apachectl[2654]: Action 'start' failed.
Oct 31 23:05:27 raspberrypi3 apachectl[2654]: The Apache error log may have more information.
Oct 31 23:05:27 raspberrypi3 systemd[1]: apache2.service: Control process exited, code=exited status=127
Oct 31 23:05:27 raspberrypi3 systemd[1]: Failed to start The Apache HTTP Server.
Oct 31 23:05:27 raspberrypi3 systemd[1]: apache2.service: Unit entered failed state.
Oct 31 23:05:27 raspberrypi3 systemd[1]: apache2.service: Failed with result 'exit-code'.
Hint: Some lines were ellipsized, use -l to show in full.
Processing triggers for systemd (232-25+deb9u4) ...
Installing phpMyAdmin
Reading package lists... Done
Building dependency tree
Reading state information... Done
phpmyadmin is already the newest version (4:4.6.6-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
END: Packages

kevinkey619, 2018-11-01:
I uninstalled apache2, uninstalled Acuparse, and then reinsalled.  Now this is what I get at the homepage

<?php
/**
 * Acuparse - AcuRiteÂ®â€Ž Access/smartHUB and IP Camera Data Processing, Display, and Upload.
 * @copyright Copyright (C) 2015-2018 Maxwell Power
 * @author Maxwell Power <max@acuparse.com>
 * @link http://www.acuparse.com
 * @license AGPL-3.0+
 *
 * This code is free software: you can redistribute it and/or modify
 * it under the terms of the GNU Affero General Public License as published
 * by the Free Software Foundation, either version 3 of the License, or
 * any later version.
 *
 * This program is distributed in t

Is the designed to work with Raspberry Pi?

maxwellpower, 2018-11-01:
Having the code displayed means that PHP is not working properly.

To see what is going wrong, it would be good to start with a fresh install of the OS then work through the installation steps manually.

closed 2018-11-06 by maxwellpower