acuparse/acuparse #57: Help With Set

csmithsound, 2019-03-07:


csmithsound, 2019-03-07:
Is there any instructions on the setup once in the browser? 
need some basic pointers. 
I cant get to live data/archive yet.

Specifically: 

1. 5-in-1 Station ID
- Where do i find this? my Station ID on weather underground is a different format and isn't accepted (previously my 5in1 was reporting to wunderground via the acurite server)

2.mac address of smart hub
- i assume its still the one on the sticker on base of hub?

3.Database Settings 
- i assume i got this correct as it allowed me to proceed to next page
hostname: localhost
database: acuparse
username: acuparse
password: mypassword

4. Site Settings hostname - 'localhost/Pi.ip.address' ? default was 'localhost'

5. Site Settings Email - do i just stick to the default? i think i skipped the email bit in the installation as didn't understand it. 

Help will be much appreciated. 

Chris.

P.S: thank-you for making this software! i had only just purchased my 5in1 and smart hub when Acurite announced they were going to pulled the plug.

maxwellpower, 2019-03-07:
I believe a lot of these are covered in the install guide.

1. Your 5N1 ID is on your 5N1 same as the SmartHub. You can find this in your /var/log/syslog, if your DNS redirect is set up properly.
2. SmartHub MAC is found on the SmartHub.
3. You set up the database credentials as part of the installation script. It asks you to choose these and so these would be the values you set.
4. The hostname is either the FQDN or the IP address of your system.
5. Email is not important but you must set something if you want emails to send.

Hope you are able to get it up and running with your data :)

closed 2019-03-07 by maxwellpower

csmithsound, 2019-03-07:
Great thanks, very new to this. how do i find /var/log/syslog?

pi@raspberrypi:~ $ /var/log/syslog
-bash: /var/log/syslog: Permission denied
pi@raspberrypi:~ $ sudo /var/log/syslog
sudo: /var/log/syslog: command not found

maxwellpower, 2019-03-07:
`tail -f /var/log/syslog`

csmithsound, 2019-03-07:
Its working! Thanks :-)