acuparse/acuparse #58: Remove Acurite updates for SmartHub

maxwellpower, 2019-03-07:
Acurite has retired the SmartHub for good. Time to remove updates to MyAcurite and cleanup updates page.

pvince, 2019-03-07:
Is it possible to translate the smartHub requests to use the access API?

maxwellpower, 2019-03-08:
Most definitely! I'd say that this is a moral/permission dilemma, more than anything.

One of the many reasons Acurite canned the SmartHub would have been because the updates were too frequent and that has a cost at scale. Either way, a good implementation would run the same way CWOP updates do and so could be done, with permission.

I would assume that Acurite wants the cash users upgrading to an Access brings. This makes me think they might be against this, if only for that reason.

I am of course open to a discussion on if this should be done instead of removing support completely.

mfrazzz, 2019-03-08:
If you were to do this, I would say that Acuparse would need to send the
data like an Access would (no more than once every 5 minutes or whatever it
is). The other issue would be registering the device to MyAcurite as you
have to give the hardware id don't you? My guess is a SmartHub address
wouldn't work. Then if you get into spoofing an address there are
definitely even more moral implications to consider.

maxwellpower, 2019-03-08:
### TL;DR: If I'm going to buffer any updates, I would rather it be real-time Access updates for upload to MyAcurite every 5 mins.

Is a good point @mfrazzz I did not totally consider the hardware ID of the Access; only the 5N1. The Access would send it's MAC and that would be the real issue. The only way to know if it would accept a SmartHub MAC would be to try it. Something I am a little squeamish about since it's their service and not mine. Spoofing an Access MAC or re-using the same one would be nothing but problems. 

Honestly, if anything, I would prefer real-time updates out of the Access; in a responsible way of course. Only really need to buffer them when there is a power outage, then just replaying them all. All they would really need to do is allow a custom server without the need for any workaround. If the upload hostname is not theirs, provide real-time. End users who just want MyAcurite won't change this. Ultimately it would save them bandwidth on their servers. I suspect they could be mulling a subscription-based SaaS model. This project does not help them with that in any way nor would it hurt them based on what I believe the current user count to be (<1000, I don't have any good way to track). It is, of course, growing fast, ever since they ended support for the SmartHub. See this weeks GH stats below:
[image]

closed 2019-03-16 by maxwellpower

garyt72, 2019-03-19:
Do you know if there is any way to increase the Access reporting time?  I would really like to see updates more frequently than every 5 minutes?  I would be willing to turn off uploads to MyAcurite all together as a tradeoff.

maxwellpower, 2019-03-24:
I agree with this Gary! I have been hammering home this point since the Access came out. Right now, I have not been able to find a way to increase the reporting time. I would love Acurite to allow us to do this. Unfortunately, I don't think they care at all about us.

commit d8b2ca5b (2019-03-07): Removing support for uploading SmartHub to MyAcurite
Closes #58
PR #61 (2019-03-16): Removing support for uploading SmartHub to MyAcurite
Closes #58
commit b5c3fa81 (2019-03-16): Removing support for uploading SmartHub to MyAcurite
Closes #58