acuparse/acuparse #59: Deploy Acuparse via Docker image

maxwellpower, 2019-03-07:
Create a Docker image for general use.

kasperwire, 2019-04-09:
Great product you have here.  I noticed creating a docker image is on your todo list.  I run UNRAID and would love to be able to add this to a docker on my system.  Looking forward to release 3 and testing this out.

dijikul, 2019-04-16:
Containerization of Acuparse is greatly desired!  Consider me eager for your updates!

maxwellpower, 2019-04-23:
I've been working on the best way to deploy and package out. It's still in progress. There are two docker builds required. One for X86 and one for ARM. The ARM image is a little more difficult. So far, I always find myself requiring a dedicated machine. That is most likely the model.

To deploy with docker we will most likely still need a VM host, then use Docker Compose to build the environment.

Running all of Acuparse inside one container for production seems like a terrible idea. I don't know what I would have as PID1 and maintain that container. If anything, I will use a single container for testing and development and compose the rest for production.