acuparse/acuparse #60: edit the database

csmithsound, 2019-03-07:
On initial set up the temperature was recorded at -17C when actual temperature was 4C. After a few seconds the temperature corrected. 

Is there anyway i can edit the database to remove these first few readings? 

[image]

maxwellpower, 2019-03-07:
This should do it:
`mysql -u<USERNAME> -p<PASSWORD> -e "DELETE FROM acuparse.archive WHERE tempC = -17"`

csmithsound, 2019-03-07:
great thanks, i actualy realised i could juse re create the database.

csmithsound, 2019-03-07:
-17.8 came back after deleting and recreating. This time it wasn't there initially. 

pi@raspberrypi:~ $ sudo mysql -uroot -pchr1st0ph3r -e "DELETE FROM acuparse.archive WHERE tempC = -17.8”
> 

that command 
doesn't seem to work.

mfrazzz, 2019-03-07:
Do you get an error or status back after the command runs?

maxwellpower, 2019-03-08:
Since -17.8C is a rounded 0F try running:
`mysql -uroot -p<PASSWORD> -e "DELETE FROM acuparse.archive WHERE tempF = 0"`

tempF is the reading received from the sensors.

csmithsound, 2019-03-08:
after the comand i get '>' 

pi@raspberrypi:~ $ sudo mysql -uroot -pmypassword -e "DELETE FROM acuparse.archive WHERE tempC = -17.8”
>

csmithsound, 2019-03-13:
ok so i found out that means the command is still running but i left it all day and it didn't complete. 
How long should this take? 
can i just delete the timeframe instead? 
Low: -17.8℃ (0℉) on 7 Mar @ 17:24

maxwellpower, 2019-03-13:
Try
`mysql -uroot -p<PASSWORD> -e "DELETE FROM acuparse.archive WHERE tempF = 0;"`

GaryKOHOHIO2, 2019-03-15:
csmith, I THINK you need a semi-colon at the end of sql commands. I'm inexperienced, but I frequently forget that.

closed 2019-03-24 by maxwellpower

csmithsound, 2019-04-10:
this one worked. ;-)

sudo mysql -uroot -pchr1st0ph3r -e "DELETE FROM acuparse.archive WHERE tempF = 0";