acuparse/acuparse #62: Atlas Station ID

MCancellier, 2019-03-08:
Where do I find the Station ID for an Atlas? I looked all over the sensor for some kind of label and didn't find anything.

mfrazzz, 2019-03-08:
If your Atlas is running, ssh into your Acuparse box and do a:
tail -f /var/log/syslog

You can then look for an Atlas and it will have the ID.

MCancellier, 2019-03-08:
I’m still in the setup process. The dusk of doesn’t have any data yet.

maxwellpower, 2019-03-13:
The Access will need to have a DNS redirect setup or you will need to use the included script to change the hostname that updates are sent too from the Access. You should be able to complete setup after creating your user and entering the database information. Then go to admin and choose Upload Server under Access Tools.

closed 2019-03-13 by maxwellpower