acuparse/acuparse #63: Data access API

onato, 2019-03-14:
I'm really liking Acuparse for getting my data up to weather cloud. I'd like to create some "flashy graphics" using a json feed directly from Acuparse. I see that the have what looks like the current days data at `/?json`.

I'd like to create something like the following (from weathercloud) with a date range selector. My goal is to see from where the predominant wind comes from at a particular location and how that changes througout the year.

[image]

To do so I'd need to be able to fetch data similarly to…

/daily/wind?from=2017-01-01&to=2018-01-01

This would return an array with the wind data for each day in the date range.

Are there any plans to provide something like this? Otherwise, I'd take a look at how I can get the data directly out of MySQL.

maxwellpower, 2019-03-16:
Is something I would consider. Don't know how long something like this would take to implement. There would obviously need to be more endpoints than just wind and there would need to be quite a few data points to search.

onato, 2019-03-17:
I would like it if acuparse would focus on collecting and providing the data. 

If it would be easy to get the data out through a JSON API, I'm sure you would find plenty of people implementing interesting ways to display the data they are interested in. You could then add the good ones to the acuparse dashboard.

Perhaps an easy way to get started would be an API to access the `archive` table?

maxwellpower, 2019-03-17:
I agree. A proper flow for data output would need to be sorted. The Archive table would be the best place to pull from since it's the only table that does not get trimmed. I would be interested in hearing about endpoints we would want to see available.

Thinking about your wind data request. I imagine such a request to the API would be intensive. When pulling a wind range. There is an archive entry for each min. Are we pulling the result from each min over a year? Or, are we looking for a max each day etc?

onato, 2019-03-18:
How about preprocessing the data by hour, day and week. Depending on the timeframe you want, you could then access the more or less granular aggregation. So, for each field, you would average the values of that particular period. We could then request the weekly data over a year, or the hourly data over a week without overheating anyone's raspberry pi.

maxwellpower, 2019-03-24:
Soon as I have some spare cycles, I will play with this and see if I can get an iteration out of it. I like the idea of an API to pull data out.