acuparse/acuparse #64: apache error in archive view

GaryKOHOHIO2, 2019-03-14:
Each time I use "archive view", it generates an apache error; undefined variable windDIR in getArchiveWeatherData.php on line 581. It is the return after 'switch' windDEG to human readable, that looks like this: 
580     }
581        return $windDIR;
582    }
Did my install miss a string or variable definition, or just a case of archived wind direction is meaningless. I don't care about the data, just cleaning up errors.
     Thank you,
          Gary

maxwellpower, 2019-03-16:
I don't get this error at all when running an archive. I suspect that you have a value stored for windDEG that does not match the switch case. You could run an SQL query to find any values that do not match the Acurite factory values.

maxwellpower, 2019-03-16:
I am going to add a default to the case for the next version that will show the raw DEG value. This should assist with finding errors when they happen and remove the error from your error log.

When running the archive, you will begin seeing the raw DEG value in RED when it is not part of the case.

commit acc00bc6 (2019-03-16): Resolve windDIR errors in archive

closes #64
closed 2019-03-17 by maxwellpower