acuparse/acuparse #68: Not Receiving date from Access

RedlineMods, 2019-03-18:
I am running debian 9.8 stretch and 2.8.0 Acuparse.  Install went fine Access was updated and now points to server by IP.  I know this is something that is probably a simple fix.  I just can't see it.  Hoping you might be able to help out.  Below is what I get when I look at the syslogs.

root@WXServ:~# tail -f /var/log/syslog
Mar 18 08:34:01 WXServ CRON[3103]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
Mar 18 08:34:02 WXServ acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB.
Mar 18 08:34:45 WXServ PackageKit: refresh-cache transaction /3289_eebeedcb from uid 1000 finished with failed after 1681ms
Mar 18 08:34:49 WXServ PackageKit: get-updates transaction /3290_cdeeceec from uid 1000 finished with success after 1650ms
Mar 18 08:35:01 WXServ CRON[3376]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
Mar 18 08:35:01 WXServ acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB.
Mar 18 08:36:01 WXServ CRON[3384]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
Mar 18 08:36:01 WXServ acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB.
Mar 18 08:37:01 WXServ CRON[3395]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
Mar 18 08:37:01 WXServ acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB

RedlineMods, 2019-03-19:
Noticed something today.  The Access is getting and sending data just not through Acuparse.  I know because weather underground is getting updated every 30 seconds from the Access.  I put in my weather underground information and it started sending.  I take it out it stops.  So my issue is that the Access is on the same network as the Acuparse but somehow is not getting the data from the Access.  What am I missing?  Or did something not get configured right.  I did not see any errors on install.  I installed a fresh debian 9.8.0 stretch with KDE interface, installed Acuparse with no problems.  Set it up and it doesn't get data from the Access.

maxwellpower, 2019-03-24:
Did you use the page to set the Access host where data should be sent? If that has been done, the Access should be dumping data to your Acuparse instance. Keep in mind the Access only drops data every 5 mins, so you have to watch the wire a bit.

If you have done that and are not getting updates in the syslog, there is a network issue somewhere. You can try a Wireshark and see if the packets are running along your network.

The WeatherUnderground updates are a separate piece and are not related to the data we are looking for. When you get Acuparse working, I would recommend disabling your updates to WU as it COULD cause issues with double readings. It's probably less trouble with the Access because of 5 min updates though.

RedlineMods, 2019-03-25:
Okay packets are exiting fine from my network back to acurite.  When I go to change the access update server, are we allowed to put in an IP address or do we need to have a host name?  When I put in the IP address of the server that is on the local network that runs acuparse it stops working.

maxwellpower, 2019-04-18:
I can't say for sure since I did not write the firmware for the Access but I would assume IP is just as good as a hostname. Although, I use a hostname myself.

closed 2019-05-17 by maxwellpower