acuparse/acuparse #69: "l" or "1" after install

MoonFri, 2019-04-17:
After installing the first time and you are stepping up your database. You may need to click on the Green X before adding your information to the Initial Database Setup.

I was receiving errors and needed to continuously reset the configuration with 

sudo rm /opt/acuparse/src/usr/config.php

once I closed the green information header before inputting the date for the Initial Database Setup it worked fine for me. 

Frustrating to finely find the answer. Hopefully this advice helps.

dijikul, 2019-04-18:
This issue is affecting me also.

removing the config.php as you suggest allows me to return to the initial configuration menu.  Is the "green information header" you're referring to, the popup that appears with the text, 'Database Configuration saved successfully!' after the first load?  

Your suggestion gave me hope, but doesn't appear to be working.

maxwellpower, 2019-04-18:
How are you initially installing? Debian, Rasberry?

MoonFri, 2019-04-18:
I installed Raspbian Stretch Lite Headless and followed the instructions on

https://github.com/acuparse/acuparse/wiki/Installation-on-Raspberry-Pi

Yes the green header was the pop up. Close it right when you open the webpage before entering any information like localhost and passwords. To be safe close the all pop ups before entering anything as a precaution. 

I will try to go through the setup again with another Pi and Card this weekend. If successful happy to upload and image if it helps.

MoonFri, 2019-04-18:
The Initial Database Settings screen should have the dialog pop up closed before entering 

Hostname
Database
Username
Password 

Then Save Settings. 

I used defaults except for username and password.

dijikul, 2019-04-23:
I'm trying to get things running in docker. Lots of dependencies missing that aren't listed (to be expected).  Today I learned, however, that the Weather Underground API is end of life (per https://apicommunity.wunderground.com/weatherapi/topics/end-of-service-for-the-weather-underground-api).  Will Acuparse be updated to work with the new PWS API?

Thanks again for your work.

maxwellpower, 2019-04-23:
@dijikul Answered in #70

MoonFri, 2019-05-16:
I ran into another issue with the dreaded "l" or "1" after trying to change my password in 

http://mysite.com/admin/account?password. 

Again on a Raspberry Pi. I removed my config.php file again and started again. I did not lose my old data in the mySQL database. 

I now have a backup of the config.php in the same directory named "config-backup.php" and will experiment again. I needed to get back up and running quickly as rain in CA doesn't come very often and didn't want to miss the rain collection recording.

maxwellpower, 2019-05-17:
I have a suspicion that this is related to the way the database is created on the PI specifically. It might be an issue with package versions. At some point here I will pick up a PI and test it. Right now, PI is not "officially" supported. I think the installer just needs to be modified to ensure the proper packages/commands needed run instead of the same ones as Debian/Ubuntu.