acuparse/acuparse #71: Windy.com API support request

gus70, 2019-05-03:
This is a request to add Windy.com upload capability using their new API.
https://community.windy.com/topic/8168/report-you-weather-station-data-to-windy

Thanks.

mfrazzz, 2019-05-04:
According to the comments, if you are sending to CWOP, Windy is already injesting your data via MADIS.

gus70, 2019-05-04:
Yes but it isn't the same thing. There are advantages of sending directly to Windy. Your data on Windy via CWOP that comes from MADIS is delayed an average of 10 minutes and up to 19 minutes at times. When sending directly to Windy you data is never more than 5 minutes delayed. Also only when you send directly does your station show forecast precipitation with the forecasting models. Besides it sounds like Windy is developing even more features for their future API versions. 

See here for details and comparison.
https://community.windy.com/topic/8263/windy-station-vs-cwop-station

maxwellpower, 2019-05-04:
I think this should be easy enough to implement either way. I think there is going to be a market shift over time away from WU as they continue to lock down their API. I believe it's best to support as many 3rd party providers as possible.

commit 7294219a (2019-05-18): Add Windy Support

Closes #71
commit e3715901 (2019-05-18): Add Windy Support

Closes #71
commit 7126921c (2019-05-18): Add Windy Support

Closes #71
commit 5712785c (2019-05-19): Add Windy Support

Closes #71
closed 2019-05-19 by maxwellpower