acuparse/acuparse #73: Not receiving data

MCancellier, 2019-05-16:
Hi,
I’m new to Acuparse. I just built a Raspberry Pi 3B+ and installed Acuparse for use with an Acurite Access and Atlas. Once I completed the setup and ran the script to repoint the Access to the Pi, I cannot seem to get any kind of data.

Are there any simple things I can check right off the bat?

Thanks!

maxwellpower, 2019-05-17:
Look at your syslog and see if there are any errors there. Ensure the Access is on the same network and that they are able to talk to each other.

`tail -f /var/log/syslog`

nneelix, 2019-05-17:
I had similar issues after install on Pi/Ubuntu.  Was able to figure out that Acuparse was not  creating/connecting to the DB due to permissions.  Manually created the DB via provided manual instructions and the '1' went away.  However I could still not run the setup or create initial user.  Using myphpadmin I noticed that db still had no tables and/or data, used my myphpadmin to run the SQL provided in github and everything has worked since.

MCancellier, 2019-05-17:
It’s always the same:

May 16 20:10:01 AcuParse CRON[1577]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
May 16 20:10:02 AcuParse acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB.
May 16 20:11:01 AcuParse CRON[1605]: (root) CMD (php /opt/acuparse/cron/cron.php > /opt/acuparse/logs/cron.log 2>&1)
May 16 20:11:01 AcuParse acuparse(2.8.0-release): (SYSTEM)[ERROR]: OFFLINE: not receiving data from the Access/smartHUB.

The Access and the Pi are on the same network connected via ethernet. I can ping the Access from the Pi. The Access was configured via the script to look at the Pi’s IP. I’ve tried this with two different Access units and two different Pi’s. I don’t know what else to try.

maxwellpower, 2019-05-17:
@nneelix I think you meant to reply in #69

maxwellpower, 2019-05-17:
This has come up before and it's possible the Access expects the hostname to be a FQDN and not an IP address. If you have a local domain that can resolve, it would be best to try that. Maybe even just the text hostname of your PI?

nneelix, 2019-05-17:
I'm using IP address without DNS redirect on my setup without issue.  Instead I used a dhcp reserve for the Access and then added the IP address of my Access to <acuparse>/admin/access.  NOTE:  The access ip address is great location to get sensor id for acuparse setup.