acuparse/acuparse #75: Battery Indicator -

MoonFri, 2019-06-01:
I think it may be helpful to have the feature of a battery status.

maxwellpower, 2019-06-02:
The data is sent with the readings but it is not saved in the database.

Where do you see this status being displayed?
I think if I was to keep the data in the DB, would then create a new admin page, to show the tower status. Maybe in the view Tower screen as a separate column?

MoonFri, 2019-06-06:
With the 5-1 and SmartHub you can look at the status page and see the battery status. Using Acuparse you need to review the logs(see below) to see the battery status or plug the SmartHub directly in to the network to review the status.  

Jun  5 19:00:40 raspberrypi acuparse(2.9.0-release): (HUB)[MyAcuRite]: Query = dateutc=now&action=updateraw&realtime=1&id=24C86F0AEEA5&mt=5N1x31&sensor=00002250&windspeedmph=1&winddir=270&rainin=0.00&dailyrainin=0.00&baromin=29.88&**battery=normal**&rssi=3 | Response = {"localtime":"19:00:40"}

It would be a nice to have enhancement just for when viewing the Acuparse Status page not for forwarding to external websites. 

I have had battery issues and tend to track it more, recently seeing to clean the 5-1 capacitor which shorts and drains power for batteries lasting 2 months at most. Hopefully the cleaning worked and i can get longer times between replacement.